Xinyuan-LilyGO/LilyGo-EPD47 #117: How to connect the JTAG espressif esp-prog T5 epaper board

KlaasVortex, 2024-04-26:
How do I connect the JTAG to the T5 epaper board ?

I think with a much bigger picture of the Product Pin Diagram I can look it up. 
The raspberry pi connector is unreadable.

ced-mos, 2024-10-13:
@KlaasVortex Do you know, if the debugging via the integrated USB JTAG can also be activated in VS Code with Platform IO?

lewisxhe, 2024-10-14:
You can refer to this, also for ESP32S3
https://rowa.tech.blog/2024/10/04/esp32-s3-debugging-via-usb-jtag-interface-in-vscode-platformio/

ced-mos, 2024-10-16:
@lewisxhe Thank you. The guide helped me to get started. At the moment I had additionally to add `debug_tool = esp-builtin` in the `platformio.ini` in order that the debugger can connect to the device. However, currently the debugger does most of the time not halt at a break point, and if does, then at a wrong position or it does not continue, respectively halts a the same line when I execute continue.

Here the extract of my c code, where it always breaks a line `esp_task_wdt_init(120, false);`.
```
void setup() {
    // Disable Task Watchdog Timer
    esp_task_wdt_init(120, false);

    ... other code
```

The output of the debug log is

```

Thread 
1 "loopTask" received signal SIGTRAP, Trace/breakpoint trap.
2
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
0x40374409 in ?? ()
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
Info : [esp32s3.cpu0] Target halted, PC=0x420694D6, debug_reason=00000000
[esp32s3.cpu0] Target halted, PC=0x420694D6, debug_reason=00000000
Info : Set GDB target to 'esp32s3.cpu0'
Set GDB target to 'esp32s3.cpu0'
Info : [esp32s3.cpu1] Target halted, PC=0x40374409, debug_reason=00000001
[esp32s3.cpu1] Target halted, PC=0x40374409, debug_reason=00000001

Thread 
1 "loopTask" received signal SIGTRAP, Trace/breakpoint trap.
Info : [esp32s3.cpu0] Target halted, PC=0x403B2482, debug_reason=00000001
0x40374409 in ?? ()
```

Do you have any guess what it could be? Also, I am using a mac with version 12.7.6 (21H1320).

closed 2024-12-01 by github-actions[bot]