Xinyuan-LilyGO/LilyGo-EPD47 #118: Cant run the demo code on my T5-S3 4.7

3eppo, 2024-06-04:
Hello,
i recently bought the T5-S3 4.7" (https://de.aliexpress.com/i/1005004647326743.html), but couldn't get it to work:
- I installed the arduino-esp32 package and updated it to th version 2.0.3
- I choose ESP32 Dev Module
- I enabled PSRAM
- I loaded the demo

The code doesn't compile: `C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino: In function 'void setup()':
C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino:109:40: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
  109 |     writeln((GFXfont *)&FiraSans, buf, &cursor_x, &cursor_y, NULL);
      |                                        ^~~~~~~~~
      |                                        |
      |                                        int*
In file included from C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino:6:
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src/epd_driver.h:368:64: note:   initializing argument 3 of 'void writeln(const GFXfont*, const char*, int32_t*, int32_t*, uint8_t*)'
  368 | void writeln(const GFXfont *font, const char *string, int32_t *cursor_x,
      |                                                       ~~~~~~~~~^~~~~~~~
C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino:109:51: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
  109 |     writeln((GFXfont *)&FiraSans, buf, &cursor_x, &cursor_y, NULL);
      |                                                   ^~~~~~~~~
      |                                                   |
      |                                                   int*
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src/epd_driver.h:369:23: note:   initializing argument 4 of 'void writeln(const GFXfont*, const char*, int32_t*, int32_t*, uint8_t*)'
  369 |              int32_t *cursor_y, uint8_t *framebuffer);
      |              ~~~~~~~~~^~~~~~~~
C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino:113:44: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
  113 |     writeln((GFXfont *)&FiraSans, string1, &cursor_x, &cursor_y, NULL);
      |                                            ^~~~~~~~~
      |                                            |
      |                                            int*
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src/epd_driver.h:368:64: note:   initializing argument 3 of 'void writeln(const GFXfont*, const char*, int32_t*, int32_t*, uint8_t*)'
  368 | void writeln(const GFXfont *font, const char *string, int32_t *cursor_x,
      |                                                       ~~~~~~~~~^~~~~~~~
C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino:113:55: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
  113 |     writeln((GFXfont *)&FiraSans, string1, &cursor_x, &cursor_y, NULL);
      |                                                       ^~~~~~~~~
      |                                                       |
      |                                                       int*
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src/epd_driver.h:369:23: note:   initializing argument 4 of 'void writeln(const GFXfont*, const char*, int32_t*, int32_t*, uint8_t*)'
  369 |              int32_t *cursor_y, uint8_t *framebuffer);
      |              ~~~~~~~~~^~~~~~~~
C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino:117:44: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
  117 |     writeln((GFXfont *)&FiraSans, string2, &cursor_x, &cursor_y, NULL);
      |                                            ^~~~~~~~~
      |                                            |
      |                                            int*
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src/epd_driver.h:368:64: note:   initializing argument 3 of 'void writeln(const GFXfont*, const char*, int32_t*, int32_t*, uint8_t*)'
  368 | void writeln(const GFXfont *font, const char *string, int32_t *cursor_x,
      |                                                       ~~~~~~~~~^~~~~~~~
C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino:117:55: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
  117 |     writeln((GFXfont *)&FiraSans, string2, &cursor_x, &cursor_y, NULL);
      |                                                       ^~~~~~~~~
      |                                                       |
      |                                                       int*
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src/epd_driver.h:369:23: note:   initializing argument 4 of 'void writeln(const GFXfont*, const char*, int32_t*, int32_t*, uint8_t*)'
  369 |              int32_t *cursor_y, uint8_t *framebuffer);
      |              ~~~~~~~~~^~~~~~~~
C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino:121:44: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
  121 |     writeln((GFXfont *)&FiraSans, string3, &cursor_x, &cursor_y, NULL);
      |                                            ^~~~~~~~~
      |                                            |
      |                                            int*
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src/epd_driver.h:368:64: note:   initializing argument 3 of 'void writeln(const GFXfont*, const char*, int32_t*, int32_t*, uint8_t*)'
  368 | void writeln(const GFXfont *font, const char *string, int32_t *cursor_x,
      |                                                       ~~~~~~~~~^~~~~~~~
C:\Users\black\AppData\Local\Temp\.arduinoIDE-unsaved202454-7196-tmplbp.39ca\demo\demo.ino:121:55: error: invalid conversion from 'int*' to 'int32_t*' {aka 'long int*'} [-fpermissive]
  121 |     writeln((GFXfont *)&FiraSans, string3, &cursor_x, &cursor_y, NULL);
      |                                                       ^~~~~~~~~
      |                                                       |
      |                                                       int*
[... 60 lines cut ...]
      |     ^~~~
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src\ed047tc1.c:58:5: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src\ed047tc1.c: In function 'fast_gpio_set_lo':
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src\ed047tc1.c:63:5: error: 'GPIO' undeclared (first use in this function)
   63 |     GPIO.out_w1tc = (1 << gpio_num);
      |     ^~~~
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src\i2s_data_bus.c: In function 'gpio_setup_out':
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src\i2s_data_bus.c:450:21: error: 'GPIO_PIN_MUX_REG' undeclared (first use in this function)
  450 |     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
      |                     ^~~~~~~~~~~~~~~~
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src\i2s_data_bus.c:450:21: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\black\Documents\Arduino\libraries\LilyGo-EPD47-master\src\i2s_data_bus.c:450:45: error: 'PIN_FUNC_GPIO' undeclared (first use in this function)
  450 |     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
      |                                             ^~~~~~~~~~~~~
Multiple libraries were found for "SD.h"
  Used: C:\Users\black\AppData\Local\Arduino15\packages\esp32\hardware\esp32\3.0.0\libraries\SD
  Not used: C:\Users\black\AppData\Local\Arduino15\libraries\SD
exit status 1

Compilation error: exit status 1`

Please help me use this board.

Thanks and Regards
B

lewisxhe, 2024-06-08:
[image]

lewisxhe, 2024-06-08:
It is best to use versions 2.0.14 and 2.0.15. Anything higher than this will definitely result in an error.

asg02fever, 2024-07-01:
i have the same board , i had the 2.0.15 versiom, all set up , not working can you post a bin test fil of it to be sure the baord isnt borken

pbenaim, 2024-10-27:
Hello everyone,

Help me!!!!

epd_init(); crashes the ESP32-S3.

I have to unplug it and press the Boot button while plugging it back in to be able to access it again. 
Pressing RST does nothing anymore not even a reboot!!!

I tried with many Arduino ESP32 2.0.14 2.015 and more ( Ver above and under )...

Drivers come from here ( https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/esp32s3/src )

For those who would tell me, your device does not work.

It worked, until the day I wanted to change my software to improve it...

So, it is not improved since it is unusable... LOL

Could there be another library that is causing this problem ?

I tryed with Arduino and VSCode ( PlatformIO ) same problem :-(

lewisxhe, 2024-10-28:
Can you provide the log that caused the crash? If it is not started, confirm whether you can find the ESP port in the device manager. If the port exists, then you can definitely upload the sketch.

pbenaim, 2024-10-28:
No log, I can compile and upload on it
When esp32 start I have serial.print juste before epd_init() and after the esp32 release the port and after that impossible to reboot with RST button 😭

I must unplug and push the Boot button and plug again to have access to ESP32. 😜

Any other ideas?

G6EJD, 2024-10-28:
You are compiling for the ESP32-S3 and have enabled PSRAM? It's a common mistake.

pbenaim, 2024-10-28:
Yes of course

pbenaim, 2024-10-28:
Well, after some research, the problem is hard, not soft.

I think the problem comes from the converter that powers the screen. I discovered it by turning the device off and on several times in a row.

Sometimes (rarely), it starts fine but after a few seconds, or even a minute, everything stops again :-(

I have version 2.3, is there a hard modification that could save my life?

Thank you for your help

lewisxhe, 2024-10-29:
Can you find the port? Disconnect the battery, hold down the BOOT button, and then plug in the USB-C.

random-0110-dude, 2024-11-21:
@pbenaim try adding 470uF capacitor across 3V3 rail as suggested in https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/98

pbenaim, 2024-11-21:
Thanks I will try :-)

closed 2025-01-06 by github-actions[bot]