Xinyuan-LilyGO/LilyGo-EPD47 #120: Problem including the library downloaded from Github

ilker-aktuna, 2024-06-14:
I am following instructions to include the library on Arduino IDE.
It displays as installed but then while compiling I get these errors:
```

In file included from c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c:6:
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.h:114:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'epd_output_row'
  114 | void IRAM_ATTR epd_output_row(uint32_t output_time_dus);
      |                ^~~~~~~~~~~~~~
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.h:119:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'epd_skip'
  119 | void IRAM_ATTR epd_skip();
      |                ^~~~~~~~
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.h:124:21: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'epd_get_current_buffer'
  124 | uint8_t * IRAM_ATTR epd_get_current_buffer();
      |                     ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.h:132:16: error: expected '=', ',', ';', 'asm' or '__attribute__' before 'epd_switch_buffer'
  132 | void IRAM_ATTR epd_switch_buffer();
      |                ^~~~~~~~~~~~~~~~~
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c: In function 'fast_gpio_set_hi':
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c:58:5: error: 'GPIO' undeclared (first use in this function)
   58 |     GPIO.out_w1ts = (1 << gpio_num);
      |     ^~~~
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c:58:5: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c: In function 'fast_gpio_set_lo':
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\ed047tc1.c:63:5: error: 'GPIO' undeclared (first use in this function)
   63 |     GPIO.out_w1tc = (1 << gpio_num);
      |     ^~~~
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\i2s_data_bus.c: In function 'gpio_setup_out':
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\i2s_data_bus.c:450:21: error: 'GPIO_PIN_MUX_REG' undeclared (first use in this function)
  450 |     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
      |                     ^~~~~~~~~~~~~~~~
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\i2s_data_bus.c:450:21: note: each undeclared identifier is reported only once for each function it appears in
c:\Users\ilker\Documents\Arduino\libraries\LilyGoEPD47\src\i2s_data_bus.c:450:45: error: 'PIN_FUNC_GPIO' undeclared (first use in this function)
  450 |     PIN_FUNC_SELECT(GPIO_PIN_MUX_REG[gpio], PIN_FUNC_GPIO);
      |                                             ^~~~~~~~~~~~~
```

so what's wrong ?

lewisxhe, 2024-06-14:
Please check the readme file to reduce the ESP32 core version to below 3.0

ilker-aktuna, 2024-06-14:
thank you. downgrading to 2.0.17 solved the issue with compiling.
But after uploading the sketch  device went into boot loop.
I think that's because I used the esp32s3 version (how do I know which version I should use ?)
now I am trying to upload non-s3 version but I get :

```
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0x30): Possible serial noise or corruption.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
Failed uploading: uploading error: exit status 2

```

lewisxhe, 2024-06-15:
Please follow the README indications that the option is correct, PSRAM is set to OPI

ilker-aktuna, 2024-06-15:
> Please follow the README indications that the option is correct, PSRAM is set to OPI

There is no OPI option in the Arduino IDE. Only "disabled" and "enabled" , and I have already selected "enabled"
And I think my issue is not about PSRAM, I need toput the device in download mode.
which one is the reset and boot buttons on this device ?

ilker-aktuna, 2024-06-15:
the problem was fixed changing the JTAG adapter option on the IDE.
Setting it to USB bridge solved the problem.
Also the download mode of this device is triggered by:
press S5+S6 buttons
release S5
release S6
on serial console it says "waiting for download"

closed 2024-06-15 by ilker-aktuna