Xinyuan-LilyGO/LilyGo-EPD47 #124: Weather display stuck on july 17

ramatomo, 2024-07-29:
My weather display they suddenly stopped working
I created 2 projects, one for home and one for work, and both were stuck on July 17th and are no longer updated...

I tried to reload the sketch (https://github.com/Xinyuan-LilyGO) and check the api but nothing changed.
then I tried to load the DEMO sketch (loaded correctly) and then reload the weather display sketch, but despite it being loaded correctly the screen always shows the DEMO

G6EJD, 2024-07-29:
That’s because API has been changed significantly so all the code and json decoder will need to be revised. Also you have to subscribe to their subscription service where you get limited API calls per day and then after that start to pay for the data.

As the code is effectively a pirated copy of my original work I wish you good luck in making the amendments…

ramatomo, 2024-07-29:
> As the code is effectively a pirated copy of my original work I wish you good luck in making the amendments…

Is your work up to date and working with the new API?

G6EJD, 2024-07-29:
Oh yes, but you won’t find it. And I have spiked any other versions so much they are useless.

Nebulus64, 2024-08-12:
API 2.5 is still functional. Try using this query format : api.openweathermap.org/data/2.5/weather?lat={lat}&lon={lon}&appid={API key}. Don't use request by city name (Geocoding API).

G6EJD, 2024-08-12:
It was due for service deletion at the end of June, so it’s just a matter of time.

closed 2024-11-16 by github-actions[bot]