Xinyuan-LilyGO/LilyGo-EPD47 #127: Unable to use touch in interrupt mode

niccocco, 2024-08-31:
Hello, i'm trying to attatch a gpio interrupt to the touch_int line. But after i try to attatch an interrupt with the function `        attachInterrupt(digitalPinToInterrupt(TOUCH_INT_t), touchIsr, FALLING);
` and then try to initialize the screen, i get `Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).`

I also tried using this other method to install a gpio interrupt:
```
gpio_pad_select_gpio(TOUCH_INT_t);
gpio_set_direction(TOUCH_INT_t, GPIO_MODE_INPUT);
gpio_install_isr_service(0);
gpio_set_intr_type(TOUCH_INT_t, GPIO_INTR_NEGEDGE);
gpio_isr_handler_add(TOUCH_INT_t, touchIsr, NULL);
```
but still no luck, the program crashes after i call `epd_init();`

I know that i could be polling the pin with a task? but i wanted to use an interrupt
Anyone got this issue?

G6EJD, 2024-08-31:
Have you check the service routine is using iram and your protecting the variables on entry and exit?

niccocco, 2024-09-01:
using something like `portENTER_CRITICAL_ISR();` or i should create a mutex to limit access to the isr?

G6EJD, 2024-09-01:
Both

niccocco, 2024-09-02:
Looking at the logs sometimes i get another error: 
```
E (5361) lcd_panel.io.i80: esp_lcd_new_i80_bus(155): install interrupt failed
ESP_ERROR_CHECK failed: esp_err_t 0x105 (ESP_ERR_NOT_FOUND) at 0x4037df70
  #0  0x4037df70 in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_err.c:42

file: "src/i2s_data_bus.c" line 387
func: i2s_bus_init
expression: esp_lcd_new_i80_bus(&bus_config, &i80_bus)

abort() was called at PC 0x4037df73 on core 0

Backtrace: 0x40378996:0x3fcebb90 0x4037df7d:0x3fcebbb0 0x40383ef1:0x3fcebbd0 0x4037df73:0x3fcebc50 0x42007a8b:0x3fcebc70 0x42006f58:0x3fcebd20 0x420071be:0x3fcebd70 0x42005502:0x3fcebd90 0x4200d712:0x3fcebeb0
  #0  0x40378996 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:408
  #1  0x4037df7d in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_system.c:137
  #2  0x40383ef1 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:46
  #3  0x4037df73 in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_err.c:43
  #4  0x42007a8b in i2s_bus_init at src/i2s_data_bus.c:387 (discriminator 1)
  #5  0x42006f58 in epd_base_init at src/ed047tc1.c:140
  #6  0x420071be in epd_init at src/epd_driver.c:157
  #7  0x42005502 in setup() at code/testFunc/main.cpp:151 (discriminator 3)
```

niccocco, 2024-09-10:
Closing this issue as the problem was just an incorrect use of interrupt and freertos priorities.

closed 2024-09-10 by niccocco