Xinyuan-LilyGO/LilyGo-EPD47 #130: epd_init() seen to crash after a random number of timed awakenings from deep sleep

Dixilea, 2024-09-18:
I am encountering random crashes with my _LILYGO Screen-4.7S3 V2.3 2021-6-10_. 

I think I have pinpointed the issue to epd_init(). Here is a minimal example that suffers from crashes:

```C++
#ifndef BOARD_HAS_PSRAM
#error "Please enable PSRAM !!!"
#endif

#include <Arduino.h>
#include "epd_driver.h"
#include "utilities.h"

RTC_DATA_ATTR int boot_count{ 0 };

void setup()
{
    Serial.begin(115200);
    delay(5000);

    ++boot_count;
    Serial.println("\nBoot #" + String(boot_count) + ".");

    Serial.println("epd_init()");
    Serial.flush();
    epd_init();
    Serial.println("epd_poweron()");
    Serial.flush();
    epd_poweron();
    Serial.println("epd_clear()");
    Serial.flush();
    epd_clear();
    Serial.println("epd_poweroff()");
    Serial.flush();
    epd_poweroff_all();

    Serial.end();
    delay(1000);

    esp_sleep_enable_timer_wakeup( 10 * 1000 * 1000 ); // microseconds
    esp_deep_sleep_start();
}

void loop()
{
    // This loop is never reached. The ESP32 will go into deep sleep before it would have reached this loop.
}
```

Also, here is platformio.ini:

```ini
[platformio]
boards_dir = ../../platformio/boards

[env:t5-47-plus]
platform = espressif32
upload_protocol = esptool
framework = arduino
platform_packages =
	framework-arduinoespressif32@https://github.com/espressif/arduino-esp32.git#2.0.5
	toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
upload_speed = 921600
monitor_speed = 115200
board = lilygo-t5-47-plus
build_flags =
	-D CORE_DEBUG_LEVEL=3
	-DARDUINO_USB_MODE=1
	-DARDUINO_USB_CDC_ON_BOOT=1
	-I include
	-std=c++2a
lib_deps =
 	SPI
 	Wire
```

Here is a typical output, with the crash occurring after the 4th wake up:

```
Boot #1.
epd_init()
epd_poweron()
epd_clear()
epd_poweroff()

Boot #2.
epd_init()
epd_poweron()
epd_clear()
epd_poweroff()

Boot #3.
epd_init()
epd_poweron()
epd_clear()
epd_poweroff()

Boot #4.
epd_init()
```

I should note that the crashes occurs after a _random_ number of awakenings. Usually, the crash happens between the 2nd and 10th awakening; I don't remember it happening after the 1st. It is rare for me to see the code run successfully for more than 10 awakenings.

Any suggestions? Could the unit simply be faulty?

lewisxhe, 2024-09-19:
Is there any log output from the serial port to provide explanation?

Dixilea, 2024-09-19:
@lewisxhe Unfortunately, there is no additional output from the serial port right around a crash. Being new to ESP32s, I also don't know if it'd be possible to "extract more debug output" by enabling some feature or so.

lewisxhe, 2024-09-20:
I used this sketch and it didn't get stuck, but it looped through sleep and then restarted, you can try it

Dixilea, 2024-09-20:
I've tried the sketch you made - https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/commit/54becb5f574c4009935a00cdd2c68a78c942c671 - and can tell that the issue persists on my device. Random crashes between the 2nd and 10th awakening from deep sleep.

lewisxhe, 2024-09-21:
It seems that your device has a problem. I tested it by cycling sleep and wakeup 90 times and it didn't crash. If it is definitely a device problem, then go to the dealer.

Dixilea, 2024-09-21:
@lewisxhe Thanks for the independent test! A device problem now indeed seems like the most logical conclusion. I'll close this issue.

closed 2024-09-21 by Dixilea