Xinyuan-LilyGO/LilyGo-EPD47 #131: 4.7" Touch wont wake up from deep sleep by touching display

TheGeekFather, 2024-10-08:
It says in the docs that **you can wake up the unit by touching display**. I have tried the TOUCH examples but I don't understand how you can "wake up" the unit when poweroffall is used.

**### QUESTIONS**

1. // The touch interrupt uses non-RTC-IO, **so the touch wake-up function cannot be used to set the touch to sleep** ??????
2.  * The touch reset pin uses hardware pull-up,
    * and the **function of setting the I2C device address cannot be used.**
    * Use scanning to obtain the touch device address.*/  ???????

We have 5 of these units we want to use with homeassistant but REALLY STUCK with this issue. Saving battery drain is really important

Thank you for any help **### I REALLLLY APPRECIATE IT**

Frustrated in Canada

lewisxhe, 2024-10-08:
As described in the example, the device can only be woken up by a timer or the BOOT button.

gedger, 2024-10-29:
One option is to wire GPIO15 on the Molex connector to a touch strip/pad somewhere on your housing, GPIO15 can be used to wake from deep sleep. This means you touch the pad to wake device and then use the touch screen. Not ideal but it's the best you can do. Shame LilyGo didn't have the foresight to enable this functionality from the touch screen itself.

TheGeekFather, 2024-10-30:
Thanx gedger neat solution to waking unit. Will try it. Cheers

TheGeekFather, 2024-10-30:
Very interesting info from vroland/epdiy

[https://github.com/vroland/epdiy](url)

LilyGo Boards
There are several differences with these boards. One particular one is the way the LilyGo handles power to the display the official lilygo code has two states. This is now handled in epdiy in a different way to the lilygo code. epd_poweroff() completely turns the power off to the display and the other peripherals of the lilygo. **The new function epd_powerdown() keeps the peripherals on (this allows the touch functions to continue to work)**. epd_poweroff() should allways be called before sleeping the system You can still use touch to wake the screen with the following. In Arduino it works like this. epd_poweroff();

epd_deinit();

esp_sleep_enable_ext1_wakeup(GPIO_SEL_13, ESP_EXT1_WAKEUP_ANY_HIGH);

esp_deep_sleep_start();

closed 2024-12-14 by github-actions[bot]