Xinyuan-LilyGO/LilyGo-EPD47 #134: Touch not working?

heyudude, 2024-10-13:
Hi,

I bought this version T5 e-Paper [DE] Touch Non-Soldered Female pin [H716] in the German warehouse.
I get a reboot with touch enabled and without it displays ok.

I used this excellent example code, it uses the default touch.h
https://github.com/tommesaros/EPD-UI-Lib

Anything I missed?

Thanks for your support

lewisxhe, 2024-10-14:
What kind of touch panel are you using? There are three types, and I can't tell which driver model you are using.

heyudude, 2024-10-14:
Not sure how to check the type. It is the stock one delivered by the German office one week ago.

Is there a default program in your repositories to check the type and function? I presume also it is a driver issue.

lewisxhe, 2024-10-14:
If you just received it, you should be using the latest touch driver, try this example
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/esp32s3/examples/touch/touch.ino

lewisxhe, 2024-10-15:
The library is defined here
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/575e9539868ec6d69f13b1abe5ca4195f415a55d/platformio.ini#L52

heyudude, 2024-10-15:
Thanks! good to go!

heyudude, 2024-10-15:
thanks good to go and code!

closed 2024-10-15 by heyudude