Xinyuan-LilyGO/LilyGo-EPD47 #137: Version 2.3 Stopped Working / bricked

NorthernMan54, 2024-10-23:
I have been working with version 2.3 of this board for a few months now, and had it running for few weeks - https://github.com/NorthernMan54/esp32_balboa_spa

Then a few days ago it just stopped.  And the display is frozen at 5:11 am, and the board no longer responds.  I went and connected it to the laptop I used for development, and no serial port available.  The green and blue led does turn on, but not serial port.  I also tried the Boot/Rst step and no luck.  Also when looking via lsusb, I don't see anything.

Any clues around how to unbrick, or is it dead?

lewisxhe, 2024-10-24:
If you hold down the BOOT button and insert the USB and still can't find the port, it means it's dead.

NorthernMan54, 2024-10-24:
Tks

closed 2024-10-24 by NorthernMan54

pbenaim, 2024-11-23:
See here : https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/98