Xinyuan-LilyGO/LilyGo-EPD47 #139: #include <TouchDrvGT911.hpp>

KezohOG, 2024-10-31:
Hi there, can anyone help me on this? I have tried the recommended setup on Arduino IDE but the demo example is giving me this error.

\.arduinoIDE-unsaved2024931-20716-md7xon.wzo1n\touch\touch.ino:31:10: fatal error: TouchDrvGT911.hpp: No such file or directory
   31 | #include <TouchDrvGT911.hpp>
      |          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
exit status 1

Compilation error: TouchDrvGT911.hpp: No such file or directory

lewisxhe, 2024-10-31:
Arduino IDE -> Library manager -> Install SensorLib v0.19

KezohOG, 2024-10-31:
Thanks @lewisxhe this seems to have worked. However, there is a new error throughout the code writeln((GFXfont *)&FiraSans, "Next", &cursor_x, &cursor_y, framebuffer);

lewisxhe, 2024-10-31:
Everything compiles fine in platformio, try changing the type to int32_t
[image]

KezohOG, 2024-11-01:
Morning @lewisxhe thanks for your feedback. I took the "demo" example code and updated it to include int32_t throughout. Now there's many more errors. I'm not sure what is going on as I have yet to get one demo code working on the device and feel like I have followed all the necessary steps. Here are some of the errors. Could you provide a simple code to test/debug what is going wrong?

error: 'ESP_EXT1_WAKEUP_ANY_LOW' was not declared in this scope; did you mean 'ESP_EXT1_WAKEUP_ALL_LOW'?
  387 |                         esp_sleep_enable_ext1_wakeup(_BV(0), ESP_EXT1_WAKEUP_ANY_LOW);
      |                                                              ^~~~~~~~~~~~~~~~~~~~~~~
 
Multiple libraries were found for "SD.h"

Here is the code:
``/**
 * @copyright Copyright (c) 2024  Shenzhen Xin Yuan Electronic Technology Co., Ltd
 * @date      2024-04-05
 * @note      Arduino Setting
 *            Tools ->
 *                  Board:"ESP32S3 Dev Module"
 *                  USB CDC On Boot:"Enable"
 *                  USB DFU On Boot:"Disable"
 *                  Flash Size : "16MB(128Mb)"
 *                  Flash Mode"QIO 80MHz
 *                  Partition Scheme:"16M Flash(3M APP/9.9MB FATFS)"
 *                  PSRAM:"OPI PSRAM"
 *                  Upload Mode:"UART0/Hardware CDC"
 *                  USB Mode:"Hardware CDC and JTAG"
 *  
 */

#ifndef BOARD_HAS_PSRAM
#error "Please enable PSRAM, Arduino IDE -> tools -> PSRAM -> OPI !!!"
#endif

#include <Arduino.h>
#include "epd_driver.h"
#include "firasans.h"
#include "esp_adc_cal.h"
#include <FS.h>
#include <SPI.h>
#include <SD.h>
#include "logo.h"

#include <Wire.h>
#include <TouchDrvGT911.hpp>        
#include <SensorPCF8563.hpp>
#include <WiFi.h>
#include <esp_sntp.h>
#include "utilities.h"

#define WIFI_SSID             "Your WiFi SSID"
#define WIFI_PASSWORD         "Your WiFi PASSWORD"

const char *ntpServer1 = "pool.ntp.org";
const char *ntpServer2 = "time.nist.gov";
const long  gmtOffset_sec = 3600;
const int   daylightOffset_sec = 3600;
const char *time_zone = "CST-8";  // TimeZone rule for Europe/Rome including daylight adjustment rules (optional)

SensorPCF8563 rtc;
TouchDrvGT911 touch;

uint8_t *framebuffer = NULL;
bool touchOnline = false;
uint32_t interval = 0;
int vref = 1100;
char buf[128];

struct _point {
    uint8_t buttonID;
    int32_t x;
    int32_t y;
    int32_t w;
    int32_t h;
[... 314 lines cut ...]
                    }

                }
            }
            if (!pressButton) {
                writeln((GFXfont *)&FiraSans, buf, &cursor_x, &cursor_y, NULL);
            }

            /**
            * There are two ways to close
            * It will turn off the power of the ink screen,
            * but cannot turn off the blue LED light.
            */
            // epd_poweroff();

            /**
             * It will turn off the power of the entire
             * POWER_EN control and also turn off the blue LED light
             */
            epd_poweroff_all();
        }
    }

    delay(2);
}

lewisxhe, 2024-11-01:
Are you using the latest esp core 3.0.0 or above? This library is normal on 2.0.x. The new version has major changes and there is no time to update it. Please downgrade to 2.0.x for testing first.

This error is the definition name change of the new version. Change ESP_EXT1_WAKEUP_ANY_LOW to ESP_EXT1_WAKEUP_ALL_LOW

KezohOG, 2024-11-01:
Arduino ESP32 by Arduino 2.0.18-20240230.arduion3 is installed. Under file -> preferences -> additional board manager URLs, this is the link https://espressif.github.io/arduino-esp32/package_esp32_index.json

On line item387, I changed to ALL, new error displayed as followed:

error: 'ESP_EXT1_WAKEUP_ANY_ALL' was not declared in this scope; did you mean 'ESP_EXT1_WAKEUP_ANY_HIGH'?
  387 |                         esp_sleep_enable_ext1_wakeup(_BV(0), ESP_EXT1_WAKEUP_ANY_ALL);
      |                                                              ^~~~~~~~~~~~~~~~~~~~~~~
      |                                                              ESP_EXT1_WAKEUP_ANY_HIGH

lewisxhe, 2024-11-01:
[image]
Now 3.0.5 can also be compiled normally. Try updating

KezohOG, 2024-11-05:
Hey @lewisxhe I've tried everything, new errors. None of the example sketches are working, I've checked everything is up to date in both board and library managers. Any idea how I can get this resolved? Could you share a sample code of all the feedback thus far so I can double check the sketch is correct? Thank you for your time

[image]
[image]

lewisxhe, 2024-11-05:
[image]
When selecting the board, you may select Arduino ESP32 Boards. Remove it first, then reopen the sketch to test

closed 2024-12-21 by github-actions[bot]