Xinyuan-LilyGO/LilyGo-EPD47 #141: Lily go 4.7 Shell printed smaller than anticipated

Barnminer, 2024-11-23:
I was trying to print a shell or case for a 4.7"-S3 V2.4 from the files here: https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/esp32s3/shell

The prints came out much smaller than than the device. Are these the correct files I used?

lewisxhe, 2024-11-25:
Which file are you referring to?

Barnminer, 2024-11-28:
Sorry, https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/esp32s3/shell/style3

lewisxhe, 2024-11-28:
[image]
[image]
[image]
The size is normal

Barnminer, 2024-12-28:
I have checked measurements and looks like they are the same as the print. Are there instructions on install after printing?

closed 2025-02-12 by github-actions[bot]

jchenny7, 2025-02-25:
Could you please share what size of the screws should be used? Thank you.