Xinyuan-LilyGO/LilyGo-EPD47 #142: LilyGo weather project compilation/project(?) issue

TomasLBN, 2024-12-15:
Dears,

I was trying to run weather project, downloaded all, in ArduinoIDE all correct libraries are installed, owm_credentails filled with correct data.
There is no problem with compilation, final code is downloading to the board.
On serial terminal I'm observing something like this:
-------------------------------------------------------------------------------------------------------------------
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x4209e27e
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x44c
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a80
entry 0x403c98d0
Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400570e8  PS      : 0x00060830  A0      : 0x8200384e  A1      : 0x3fcebb20  
A2      : 0x00000000  A3      : 0xffffffff  A4      : 0x0003f480  A5      : 0x00000000  
A6      : 0x0000002b  A7      : 0x00003f48  A8      : 0x82018174  A9      : 0x3fcebaf0  
A10     : 0x00000002  A11     : 0x3c0b141e  A12     : 0x00000040  A13     : 0x00000001  
A14     : 0x000000a7  A15     : 0x3fcec720  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x400570e8  LEND    : 0x400570f3  LCOUNT  : 0x00003f47  

Backtrace: 0x400570e5:0x3fcebb20 |<-CORRUPTED
-------------------------------------------------------------------------------------------------------------------

I checked a lot of issues similar to this. 
Can anyone help me with this problem? What can I do in this case?
Thanks for any solution

Regards,
Tomasz

lewisxhe, 2024-12-16:
I can't provide more help from the log you provided. First, please stop some code to make sure the program can run correctly. Because OpenWeather updates the API, this project is no longer maintained.

G6EJD, 2024-12-16:
OpenWeatherMap is now a paid-for service so the code will no-longer operate. Even if you paid it still won't operate.

adalmolen, 2025-01-05:
> OpenWeatherMap is now a paid-for service so the code will no-longer operate. Even if you paid it still won't operate.

It still has free API (2.5) and Open Call (3.0). Open Call is free till 1000 requests a day. On registration you need to provide payment method.

Jason2866, 2025-01-15:
Small change to use API 2.5 and 3.0. Everything is working like before.
https://github.com/Jason2866/LilyGo-EPD-4-7-OWM-Weather-Display

G6EJD, 2025-01-15:
You've copied my code source and removed all my Copyright notices/licence details, either restore them or I'll issue a Copyright Strike against you on GitHub leading to this repository being removed.

Jason2866, 2025-01-15:
@G6EJD Sorry didn't noticed that where I forked removed your licence.
Readded in *.ino file and repo.
Thx for your nice work!

closed 2025-03-01 by github-actions[bot]