Xinyuan-LilyGO/LilyGo-EPD47 #143: A touch is being registered twice

sndwav, 2024-12-15:
Hello, I have the LilyGo T5 4.7" e-paper touch display and I seem to be facing an issue with it when using touch.

When I run anything that relates to touches (for example, the "demo" sketch), any touches that I make are being registered twice, meaning that I tap the screen (for example, on button B), and the line blinks and it says "Pressed Button: B", then the line blinks again and it again writes the text. 
In the "touch" example, I click "Next" and it blinks into the next page, then it blinks again and advances to the next page.

It seems to happen regardless of being connected via USB or battery.

It **doesn't** seem to happen in "demo" sketch if I tap the screen right before the voltage and date are being updated, meaning that if I tap the screen and it blinks in the touch text, and then blinks in the voltage and date updates, then there is no second blink of the touch text.

I'm using the Arduino IDE
ESP boards version 2.0.17
EPD47 version 1.0.1

Thank you for any assistance.

lewisxhe, 2024-12-16:
This is because after sending the refresh screen command, it is time to get the touch status function. Just make a small change and limit the touch detection interval to solve this problem.

sndwav, 2024-12-16:
Much appreciated! The new code examples work great. Thank you.

closed 2024-12-16 by sndwav