Xinyuan-LilyGO/LilyGo-EPD47 #144: Power consumption far too high in deep-sleep mode

Slartibatfas, 2024-12-28:
I have a Lilygo T5, 4.7 epaper display in version 2.4, which I program via Arduino. 
For a battery-powered application, I only need an update once an hour. 
Since the power consumption should be kept as low as possible, I switch off the display

_epd_poweroff_all();_

and then I put the ESP32 S3 into deep sleep mode

_esp_sleep_enable_timer_wakeup(Time_To_Sleep * 1000000ULL);
esp_deep_sleep_start();)_

However, the current consumption it still at 5-6mA and way too high for my application and far above the values specified in the data sheet for the deep sleep mode.

What can I do to keep the current consumption in deep sleep mode well below 1mA?

lewisxhe, 2024-12-30:
Test this example https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/esp32s3/examples/demo/demo.ino

Slartibatfas, 2024-12-30:
Thanks for the quick feedback! :-)
That's strange, when I install the demo, the current consumption in deep sleep mode goes down to 0.33mA (measured with Fluke 289 directly at the battery, no USB connected).
When I install my code, it stays at 6mA.

I use timed control to wake up, every hour during the day, only once at night. Here is the code:

...
  if (hour >= 6 && hour <= 22) {
  esp_sleep_enable_timer_wakeup(Time_To_Sleep * 1000000ULL);
  }
  else {
    esp_sleep_enable_timer_wakeup(Time_To_Sleep_Night * 1000000ULL);
  }
epd_poweroff_all();
WiFi.disconnect(true);
touch.sleep();
delay(100);
Wire.end();
Serial.end();
esp_deep_sleep_start();
}

I don't understand why the current consumption in deep sleep is almost 20 times higher than in the demo.ino?

lewisxhe, 2024-12-30:
Note that the timer wakeup you use has the low-speed clock turned on. You can try turning off the RTC peripheral manually. For deep sleep, see the documentation here.
```c
                        esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH,ESP_PD_OPTION_OFF);
                        esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM,ESP_PD_OPTION_OFF);
```
https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/sleep_modes.html

Slartibatfas, 2024-12-30:
I've tried that too, there were a few tips on the internet from people with the same problem, but it didn't help.
I have tried these things without any effect.

esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_SLOW_MEM, ESP_PD_OPTION_OFF);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_FAST_MEM, ESP_PD_OPTION_OFF);
esp_sleep_pd_config(ESP_PD_DOMAIN_XTAL, ESP_PD_OPTION_OFF);

I also replaced the timer wakeup  with external wakeup with an onboard button, same problem.

lewisxhe, 2024-12-30:
OK, I don't have a board right now, I will apply for a board to test, and the results should be available tomorrow.

lewisxhe, 2024-12-31:
Have you connected an external sensor? I tested the timing + external wake-up and both were 388uA. I cannot measure the 6mA you mentioned. Can you provide the minimum test code?

https://github.com/user-attachments/assets/91a7904d-5515-4282-9559-d7829b471f60

closed 2025-02-13 by github-actions[bot]