Xinyuan-LilyGO/LilyGo-EPD47 #148: Error: Unknown board ID 'T5-ePaper-S3'

DariuszJJ, 2025-01-31:
Description

When trying to build a project for the T5-ePaper-S3 board in PlatformIO, the process fails with the following error:

Resolving T5-ePaper-S3 dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing T5-ePaper-S3 (board: T5-ePaper-S3; platform: espressif8266@4.2.1; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
Error: Unknown board ID 'T5-ePaper-S3'
========================== [FAILED] Took 2.23 seconds ==========================

Steps to Reproduce

    Open VS Code with PlatformIO installed.

    Add the following configuration in platformio.ini:

    [env:T5-ePaper-S3]
    platform = espressif8266@4.2.1
    board = T5-ePaper-S3
    framework = arduino
    build_flags = -DLILYGO_T5_EPD47_S3

    Try to build the project.

Expected Behavior

The project should compile successfully for the T5-ePaper-S3 board.
Actual Behavior

The build fails with "Unknown board ID 'T5-ePaper-S3'".

amigo-6, 2025-01-31:
Hello @DariuszJJ 

try `board = lilygo-t-display-s3`. This works for me.

DariuszJJ, 2025-02-01:
Unfortunately it didn't help or is it after if it is possible please attach your platformio.ini

DariuszJJ, 2025-02-01:
Reinstalling Visual Studio Code + Platformio + deleting all APP files solved my problem

lewisxhe, 2025-02-06:
The board ID definition is here, you solved the problem, that's great
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/tree/esp32s3/boards

closed 2025-02-06 by lewisxhe