Xinyuan-LilyGO/LilyGo-EPD47 #152: Draw and write to screen epd_draw_vline vs writeln

lmediavilla, 2025-02-15:
Hello I would like to ask why epd_drvier.h if I use function to draw a line, the line is never displayed.

` epd_draw_vline(100,0,EPD_HEIGHT - 1, 256, framebuffer);`

If I add this then the line is displayed.
`epd_draw_vline(100,0,EPD_HEIGHT - 1, 256, framebuffer);
epd_draw_grayscale_image(epd_full_screen(), framebuffer);`

But then there are functions in font.c for instance this code doesn't need an epd_draw_grayscale_image or even a frame buffer to update the screen for instance:

`writeln((GFXfont *)&FiraSans, "Hello my friend", &cursor_x, &cursor_y, NULL);`

Why epd_draw_vline or any function I've tested in the epd_driver.h really draw anything without epd_draw_grayscale_image
and why writeln can write to the screen without framebuffer and any refresh function.

Is the first time I play with screens in these devices.

Kind regards.

lewisxhe, 2025-02-15:
You can go to the source code.

If the writeln function does not pass in a buffer, it will automatically apply for a buffer internally

epd_draw_vline If you pass in a buffer, it will draw in the buffer

epd_draw_grayscale_image will display the contents of the buffer

lmediavilla, 2025-02-15:
How can I make a line show on the screen directly?, if there is any way.
Sorry for me just reading the funcion definition is confusing.

lewisxhe, 2025-02-15:
There is currently no such method, you must write to the buffer and then display it

lmediavilla, 2025-02-15:
So if i want a grid of lines I need to refresh the full screen for every line?.

closed 2025-02-15 by lmediavilla