Xinyuan-LilyGO/LilyGo-EPD47 #153: Pins confusions

PaulBernier, 2025-02-27:
I have the S3 V2.4 (2024-08-28) version of the board, and after looking at [this reference image](https://cdn.shopify.com/s/files/1/0617/7190/7253/files/T5-S3V4.7_1.jpg?v=1711592882) and comparing with my board, I have a few questions:

1. Both in the board picture but also in `utilities.h` I see GPIO45 == MISO pin and GPIO48 == SCLK. But when I tested it, it was reversed, if I was using GPIO45, I could see the signal on pin marked `SCL` and when I was using GPIO48, I could actually see the signal on the pin marked MSIO (which I assume is a typo for MISO?). So isn't there a mix up here?
2. The README mentions that if not using the SC CARD, then GPIOs(16,15,11,42) are free to use. GPIO 15 and 16 are available from the left JST 4-pin connector, but it seems to me that pins 11 and 42 are not as easily accessible? Only way to use them would be to solder a wire directly on the esp32 pins?
3. The 2 JST 4-pin connectors at the bottom left, according to the [schematics](file:///home/paul/Downloads/T5-ePaper-S3-V2.4-2.pdf) the V pin (right most on the picture) is supposed to be at 3.3V, but when I measure with a voltmeter I only get 0.8V is that normal?

Thank you!

lewisxhe, 2025-02-27:
Hi PaulBernier

1. Thank you for your feedback. I have reported the pinout problem to my colleague.

2. Yes, the SD GPIO is not exposed externally. It is written in the README so that people who need it can connect it by themselves.

3. The power supply of the JST 4-pin connector must call epd_poweron();  the JST connector will be powered. It is controlled by the shift register

PaulBernier, 2025-02-27:
Thanks for the clarifications!

closed 2025-04-13 by github-actions[bot]