Xinyuan-LilyGO/LilyGo-EPD47 #154: epd_draw_image() works only if area is epd_full_screen()

michyprima, 2025-03-03:
hello, basically the title.

if I call epd_draw_image on an area which is smaller than the whole screen (for partial updates) I get garbage on the screen.

I think the calculation of the start line in the frame buffer is wrong

I'm using a T5 S3 2.4

thank you

G6EJD, 2025-03-03:
Much more likely your image format is wrong.

michyprima, 2025-03-03:
Not sure why you went to this conclusion considering my framebuffer is drawn just fine if area is 0,0,960,540

If area is smaller than that, I get stripes with the same framebuffer. 

I’m using the correct 2ppb format as you can see:

[image]

closed 2025-04-17 by github-actions[bot]