Xinyuan-LilyGO/LilyGo-EPD47 #156: partial refresh ... need more information

alainvdu69, 2025-04-04:
hi !

you say that " it cannot be partially refreshed for a long time".
what is "a long time" ? 1 hour ? more ? less ?

and how to do partial refresh : i didn't see this in the examples code.

thanks.

lewisxhe, 2025-04-07:
In the demo example, the battery voltage is updated using partial refresh.
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/blob/esp32s3/examples/demo/demo.ino

You said that partial refresh cannot be performed for a long time?

Is it what the official website said?

It can be roughly understood that after ten partial refreshes, it is best to call a full screen refresh.

alainvdu69, 2025-04-07:
hi.
ok, thanks, i will study the code.

the warning for the partial refresh is here, the blue text : https://lilygo.cc/products/t5-4-7-inch-e-paper-v2-3?variant=43808049365173