Xinyuan-LilyGO/LilyGo-EPD47 #157: Vertical Banding on Partial Refresh

AndrewLivesChaos, 2025-04-16:
I noticed that when I perform a partial refresh using epd_clear_area, the area is cleared except for two vertical bands located at the edges of the rectangle. This issue also appears in the demo. Moreover, the bands seem to be proportional to the width of the rectangle.

[issue image](https://i.ibb.co/nq3ZCtx6/Screenshot-2025-04-16-153127.png)

lewisxhe, 2025-04-17:
This seems to have always been the case. It does not affect the final display effect, right?

AndrewLivesChaos, 2025-04-17:
> This seems to have always been the case. It does not affect the final display effect, right?

Well, the pixels that belong to the bands are not cleared and therefore remain. If I have text or a value that I want to erase with a partial refresh, due to those bands the previous pixels remain because they weren't cleared.

closed 2025-06-20 by github-actions[bot]