Xinyuan-LilyGO/LilyGo-EPD47 #158: GT911 Interrupt Mode: Why Does INT Pulse Without Touch?

AndrewLivesChaos, 2025-04-17:
"I'm trying to configure the display's touch functionality. In the demo example, the coordinates are read using polling, but I would like to use interrupt mode via the INT pin, as indicated in the GT911 datasheet. With the demo example, I see a pulse of about 200 microseconds on the INT pin every now and then.

    static void Touch_Initialize()
    {
        pinMode( TOUCH_INT, OUTPUT );
        digitalWrite( TOUCH_INT, HIGH );
    
        uint8_t touchAddress = 0x14;
    
        Wire.beginTransmission( 0x14 );
    
        if ( Wire.endTransmission() == 0 )
        {
            touchAddress = 0x14;
        }
    
        Wire.beginTransmission( 0x5D );
    
        if ( Wire.endTransmission() == 0 )
        {
            touchAddress = 0x5D;
        }
        
        touch.setPins( -1, TOUCH_INT );
    
        if ( touch.begin( Wire, touchAddress, BOARD_SDA, BOARD_SCL ) )
        {
            touch.setMaxCoordinates( EPD_WIDTH, EPD_HEIGHT );
            touch.setSwapXY( true );
            touch.setMirrorXY( false, true );
            touchOnline = true;
            println( "Touch initialized" );
        }
        else
        {
            println( "Touch initialization failed!" );
        }
    }

If I set touch.setInterruptMode(FALLING), the signal stays high for a while and then the same pulse returns, but more frequently this time (I'm monitoring the signal with an oscilloscope). I don't understand whether this pulse is generated by the ESP or by the display. Is it possible to have a rising/falling edge on the INT pin only when I touch the screen?

lewisxhe, 2025-04-18:
The touch interrupt pulse is configured by the screen manufacturer. I do not recommend modifying it because I have modified the interrupt trigger mode before, which caused the touch screen to fail to obtain the correct coordinates. Because the GT911 chip is very peculiar, the manufacturer's configuration parameters can be modified directly by the user, because it must refresh all registers to effectively change, which can easily make the touch screen invalid.

The touch pulse is sent by GT911, and it is really not recommended to modify it

lewisxhe, 2025-04-18:
No, it is not necessary. As far as I remember, GT911 will have a high level trigger every 100ms. You can directly poll the register to read it.

AndrewLivesChaos, 2025-04-19:
What do you mean? Currently, with the demo example configuration, the GT911 keeps the INT signal low and generates a short positive pulse ranging from 100 to 300 microseconds. This pulse occurs approximately every 2 seconds. To achieve what you're describing, would I need to reconfigure it?

lewisxhe, 2025-04-21:
Ok, I remembered it wrong. As I said above, it is not recommended to modify it, it is very dangerous.

closed 2025-06-04 by github-actions[bot]