Xinyuan-LilyGO/LilyGo-EPD47 #161: epd_init() crashing (brownout) after deepsleep

OldPlanets, 2025-04-28:
I have the exact same issue as #130 this one with T5 (v2.4 2024-12-03) with Touch. esp_reset_reason() tells me that a brownout has occoured when I call epd_init() which happens sometimes after a wakeup from deepsleep (with all peripherals powered down before).
It seems epd_init is actually powering on some peripherals already (which I'd have expected to be in epd_poweron()), as it will turn on the blue light when called. As I understand config_reg.ep_scan_direction = true; is meant to do that. If I set it to false the blue light stays off until epd_poweron() is called.
But I suspect that just moves the problem to randomly crashing at epd_poweron() (though I am testing that right now). Any suggestions how I might fix the problem? If it's unfixable by software, would soldering a capacitor to the board help? Where would the best place for that be?

lewisxhe, 2025-04-29:
Does it happen when powered by battery alone? USB alone? Or both?

Can you take a picture of the ESP module side of the board?

OldPlanets, 2025-04-29:
It happens with both, though most of my tests I do with USB as power source.
However, after a day of testing I seem to have found a solution which works. Given that the error appears somewhat randomly and I tested just for a short time, I can't be 100% on this yet but every other way it was crashing within the first 5-10 minutes and this way it was running fine after 30 minutes (and reverting back, resulting in the same error again).
I changed config_reg.ep_scan_direction = false in epd_base_init, so not powering on the peripherals on epd_init(). Then a delay(50), then calling epd_poweron(). Without the delay it crashes too, but as I understand from the code that is the same as busy_delay(), giving the device time to actually execute the config. So initalizing it with everything off first and then turning on the peripheral seems to fix the problem for me.

I can take the pictures later if still needed (it's in a case already).

lewisxhe, 2025-04-29:
Can you take a photo so I can see if it is modified hardware?

OldPlanets, 2025-04-29:
The black connection is a 1M Ω Resistor (sensing if power source is connected)

[image]

[image]

lewisxhe, 2025-04-29:
[image]
Is this capacitor connected to the power control output?

OldPlanets, 2025-04-29:
Um, I don't know the board design (or electronics really) well enough to answer that, but there is continuity between the + side and there.

[image]

lewisxhe, 2025-04-30:
I guess it should be, I can't tell from the picture what the capacitance is.
Can you remove the capacitor first to test?
This circuit uses MOSFET to control the power supply of the peripheral. There can't be too large a capacitor in the output channel of the MOSFET, otherwise the instantaneous current at the start will be too large, causing the voltage to drop, which may cause the ESP reset trigger

OldPlanets, 2025-04-30:
Sorry, but I am in the phase of programming the device for my project and getting all the other parts connected. I can't risk to break it right now and I am not proficient enough in soldering that I am sure I can remove and reattach parts without damage.
Also as I said above it seems (again not 100% sure yet) that the initalization change in software avoids the problem for the time being.

Kowu007, 2025-05-03:
> Does it happen when powered by battery alone? USB alone? Or both?
> 
> Can you take a picture of the ESP module side of the board?

lewisxhe, 2025-05-05:
All right .