Xinyuan-LilyGO/LilyGo-EPD47 #162: problem with deleting characters

Huelva1492, 2025-05-01:
I need to erase characters on the screen without having to do a partial or total refresh, how can I do to erase a character without refreshing the screen, I need a sketch that generates a random number, erase the previous one to display the new one.

lewisxhe, 2025-05-05:
How to update the characters without partial or full refresh? This hardware can't do it

Huelva1492, 2025-05-06:
I understand that this Lilygo EPD47 e-ink board does not allow to erase a character to show another one without using the partial refresh function, I have seen e-ink display boards that change the characters without refreshing for example a clock, changing the seconds. However thanks for your answer