Xinyuan-LilyGO/LilyGo-EPD47 #164: The INT pulses kept coming out even before initializing the GT911 and did not stop after initialization.

tenchankawaii, 2025-05-19:
Hello.

I am thinking of jumpering GPIO47 to GPIO10 and using it as a wake up source for sleep.
However, I am experiencing a strange phenomenon before that.
Even before initializing the GT911, INT pulses are output forever and do not stop after initialization.
I am hampered by the problem before wakeup.
I found a similar question on the GOODIX community site, but all questions related to the GT911 were answered with “we don't deal with that here” or “ask the chip vendor”.
Since the chip vendor is supposed to be GOODIX, I could not understand what they were saying and decided that it was impossible to solve the problem on this site.
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/93#issuecomment-2488500117
Then.
>Brown: GPIO47 (47) -> MOSI (GPIO10) Optional, if you want to wake the mcu using touch
and it looks like some people have actually been able to wake it up, but I don't have any information other than the jumper.
If you know how to suppress the ghost INT pulses, please let me know.

lewisxhe, 2025-05-19:
This is the default configuration of the touch screen manufacturer. Please do not modify it. If the configuration is written incorrectly or other registers are modified, the touch screen is likely to fail. I have tried this.

I mentioned this here
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/158

tenchankawaii, 2025-05-19:
Thanks for the answer.

I had read 
https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/158 
before posting, but could not figure out why the INT pulses keep coming out.

Looking at the timing chart for “c) Gesture mode” on page 28 of the GT911 Programming Guide.pdf, it appears that INT is not output until “Issue Gesture command”.

However, in reality, the INT continues to be output before the GT911 is initialized.

I am looking for a way to stop this, but do not know the register settings.

I have created device drivers for many chips and this is the first chip that emits IRQ even before initialization.

I suspect an errata in the GT911.

tenchankawaii, 2025-05-20:
I understood that writing to register 0x8047 and beyond in GT911 is NG.

If so, what was https://github.com/Xinyuan-LilyGO/LilyGo-EPD47/issues/93#issuecomment-2488500117 
"Now, I can wake up the board using touch."

closed 2025-05-20 by tenchankawaii