Xinyuan-LilyGO/LilyGo-EPD47 #171: Device is no longer recognized as USB device

AndyTheGuardian, 2025-09-02:
Hi!
I have a problem with the LilyGo T5-4.7 Inch S3 E-paper (with Touch).
I uploaded some sample sketches without problems.
After uploading the OWM Weather station sketch in the newest version of Arduino IDE on Windows 11, the IDE crashed and since then the T5 4.7 EPD is no longer recognized (checked Arduino IDE & Windows Device Manager).
I tried to start it in Upload-Mode (by first pressing Boot (STR_I00) then Reset then letting go of first the reset button and finally the Boot-Button. But still no luck.
Both LEDs (green and blue) are on. If I press Boot + Reset while connecting to power, the blue LED is off, when letting go of the buttons it is on again.
I also tried different USB-Cables, but without success.

I'm using the 3-button EPD with the ESP32-S3 (V2.4 2024-12-03)

Thanks for your help!

lewisxhe, 2025-09-04:
1. Press and hold the BOOT button.
2. Plug in the USB.
3. Release the BOOT button. The port should display normally.

The above steps do not work with battery power.

lewisxhe, 2025-09-06:
Thank you. Please install the Serial Debug Assistant from the Microsoft Store.
https://apps.microsoft.com/detail/9nblggh43hdm?launch=true&mode=full&hl=zh-cn&gl=cn&ocid=bingwebsearch

Then follow the steps below to check if the device is in download mode.
1. Press and hold the BOOT button, connect the USB, and do not release the BOOT button.
2. Open the Serial Debug Assistant, select the port, set the baud rate to 115200, and open the port.
3. Check RTS and keep it checked.
4. Check DTR and release it. If it is in download mode, it will display:

```
rst: 0x15 (USB_UART_CHIP_RESET), boot: 0x0 (DOWNLOAD(USB/UART0))
Saved PC: 0x40041a7c
waiting for download
```

[image]

If none of the above steps work, please contact your dealer for after-sales service.